Xilinx/Vitis-AI #1412: "--fine_grained"  has some problems when runing vaitrace on KV260, vitis 3.5,

TiantianZhang, 2024-03-09:
The profiler works normally when I run python3 -m vaitrace_py --fine-grained test_accuracy_core1.py resnet5000.xmodel

xmodel is complied by vitis3.5

ERROR:root:This xmodel does not support fine grained profiling

How does one enable the fine-grained profiling on an xmodel?

Is anyone solved this problem?

The following reference is not work on KV260.

https://github.com/Xilinx/Vitis-AI/tree/v3.5/examples/vai_profiler
https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Vitis-AI-Profiler?tocId=Luv7e02rhpEpdYpzLa0cpg

same problems
https://github.com/Xilinx/Vitis-AI/issues/1279

jimheaton, 2024-03-11:
To use fine grained profiling debug mode needs to be enabled when compiling the xmodel.
Recompile the model in the vai compiler and add the following  option: --options "{'mode':'debug'}"
The default option for mode is 'normal'. Using debug mode can slightly reduce the model performance by a few percent and its recommended to set this back to normal when after you are done profiling.

jimheaton, 2024-03-11:
Yes, use: vai_q_tensorflow2 -m quantized_model.h5 -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/KV260/arch.json -o dpu_deploy.xmodel --options "{'mode':'debug'}"

TiantianZhang, 2024-03-12:
I recompile the xmodel, and run it in --fine_grained, the results are not in expectations, I'm not sure if there are any other solutions.  I have unified all versions to Vitis ai2.5

vai_c_tensorflow2 -m quantized_model.h5 -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/KV260/arch.json -o dpu_deploy.xmodel --options "{'mode':'debug'}"

**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/deploy_DPUCZDX8G_ISA1_B4096_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/workspace/rff_fpga_workspace/resnet_fpga_rff_5000/quantized_model.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 51/51 [00:00<00:00, 54154.30it/s]               
[INFO] infer shape (NHWC)  :100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 83/83 [00:00<00:00, 44608.82it/s]               
[INFO] perform level-0 opt :100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 294.54it/s]                   
[INFO] perform level-1 opt :100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 1286.00it/s]                  
[INFO] generate xmodel     :100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 83/83 [00:00<00:00, 5635.59it/s]                
[INFO] dump xmodel: /tmp/deploy_DPUCZDX8G_ISA1_B4096_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: null
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
[UNILOG][INFO] Graph name: model, with op num: 165
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/workspace/dpu_deploy.xmodel/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/workspace/dpu_deploy.xmodel/deploy.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is d4223604745d48da5f4de8069d5ec89b, and has been saved to "/workspace/workspace//dpu_deploy.xmodel/md5sum.txt"

 python3 -m vaitrace_py --fine_grained test_accuracy_core1.py deploy.xmodel

WARNING:root:The fine grained profiling model is enabled. Models will be run subgraph-by-subgraph.
1. It will interduce significant scheduling and memory access overhead, especially for small models.
2. Some xcompiler's optimizations will be turned-off.

INFO:root:VART will run xmodel in [DEBUG] mode
Analyzing symbol tables...
61 / 61
81 / 81
6 / 6
45 / 45
3 / 3
INFO:root:vaitrace compile python code: test_accuracy_core1.py
INFO:root:vaitrace exec poython code: test_accuracy_core1.py
['test_accuracy_core1.py', 'deploy.xmodel']
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0311 18:56:16.364677  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block1_conv has no mc_code
I0311 18:56:16.364840  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block2_1_conv1 has no mc_code
I0311 18:56:16.364879  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block2_1_conv2 has no mc_code
I0311 18:56:16.364918  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block2_2_conv1 has no mc_code
I0311 18:56:16.364955  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block2_2_conv2 has no mc_code
I0311 18:56:16.364997  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block3_1_conv1 has no mc_code
I0311 18:56:16.365039  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block3_1_conv2 has no mc_code
I0311 18:56:16.365082  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block3_1_shortcut has no mc_code
I0311 18:56:16.365120  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block3_2_conv1 has no mc_code
I0311 18:56:16.365159  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block3_2_conv2 has no mc_code
I0311 18:56:16.365197  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block4_1_conv1 has no mc_code
I0311 18:56:16.365236  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block4_1_conv2 has no mc_code
I0311 18:56:16.365274  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block4_1_shortcut has no mc_code
I0311 18:56:16.365314  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block4_2_conv1 has no mc_code
I0311 18:56:16.365355  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block4_2_conv2 has no mc_code
I0311 18:56:16.365394  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block5_1_conv1 has no mc_code
I0311 18:56:16.365435  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block5_1_conv2 has no mc_code
I0311 18:56:16.365476  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block5_1_shortcut has no mc_code
I0311 18:56:16.365520  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block5_2_conv1 has no mc_code
I0311 18:56:16.365562  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Block5_2_conv2 has no mc_code
I0311 18:56:16.365604  2135 dpu_kernel.cpp:178] child_name subgraph_quant_flatten_fix_reshaped has no mc_code
[... cut ...]
I0311 18:56:16.365684  2135 dpu_kernel.cpp:178] child_name subgraph_quant_Dense_dense(TransferMatMulToConv2d)_inserted_fix_2_reshaped has no mc_code
Number of RF Samples Tested is  3499
Batch Size is  1
Top1 accuracy =  0.0286
Pred Performance =  100
identification_time_per_slice=0.0006, total time = 2.0432 , total slice = 3500.0000
INFO:root:vaitrace timeout, waiting for Python thread terminated
APM Stop Collecting
INFO:root:Generating VTF
INFO:root:Processing xmodel information
INFO:root:[vart_trace.csv]: Events number: 0

jimheaton, 2024-03-12:
What results were unexpected, are you referring to the bad Top1 accuracy you are getting?

TiantianZhang, 2024-03-12:
there are errors in the log. address that there is no mc_code. if i run it without —fine_grained，there only 1 layer information and the accuracy will be normal

| |
TianTian Zhang
|
|
***@***.***
|
---- Replied Message ----
| From | Jim ***@***.***> |
| Date | 03/12/2024 23:45 |
| To | Xilinx/Vitis-AI ***@***.***> |
| Cc | TiantianZhang ***@***.***>,
Author ***@***.***> |
| Subject | Re: [Xilinx/Vitis-AI] "--fine_grained" has some problems when runing vaitrace on KV260, vitis 3.5, (Issue #1412) |

What results were unexpected, are you referring to the bad Top1 accuracy you are getting?