Xilinx/Vitis-AI #1414: Compilation failed: DPU mismatch (Cannot find target with name DPUCZDX8G_ISA0_B4096_MAX_BG2)

thallesmm, 2024-03-11:
I am having a DPU compatibility problem. I am using a ZCU104 with the DPUCZDX8G_ISA0_B4096_MAX_BG2 target, I changed the arch.json file as in https://support.xilinx.com/s/article/DPU-fingerprint-ERROR?language=en_US, from DPUCZDX8G_ISA1_B4096 to my actual target (DPUCZDX8G_ISA0_B4096_MAX_BG2) and now I am having this problem presented bellow. The DPU architecture used doesn't even appear in the registered targets. I am following the Vitis AI Tutorial: https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Introduction/03-Basic/Module_4. Should I include this architecture in the _Registry_ manually? Or even change the DPU arch of the board to the one expected (DPUCZDX8G_ISA1_B4096)? How should I do that? I am new to the Xilinx tools, so I am a bit lost in it and I would appreciate if someone could help me with that.
[image]

KrishnaGaihre, 2024-03-15:
@thallesmm if you have ZCU104 Board running with DPU image then you can run "**show_dpu**" and "**xdputil query**" command to see the DPU fingerprint directly. From those command you can get DPU fingerprint in number/values, use that one inside arch.json as mentioned [here ](https://support.xilinx.com/s/question/0D54U00006wDmkzSAC/info-post-about-dpu-fingerprint?language=en_US)and re-compile.
I also like to reconfirm, are you running Vitis AI 1.4.1 CPU/GPU docker on your Host PC? All the supported DPU architecture must be there inside Vitis AI CPU/GPU docker while using same version as in Board.

thallesmm, 2024-03-15:
Thank you for your answer.

I did exactly what you were saying. I changed the architecture in the arch.json file from DPUCZDX8G_ISA1_B4096 to DPUCZDX8G_ISA0_B4096_MAX_BG2 which is the one that appear when , however, it does not find this target.

I believe my mistake is in the 2nd part of your answer. The version of Vitis AI displayed when I run it on docker is 2.5 and the version on the board is 1.4. When pulling Vitis AI from Docker Hub I am asked by the tutorial to do the following command:

**docker pull xilinx/vitis-ai:latest**

But when setting the cross-compilation environment I downloaded the **sdk-2021.1.0.0.sh** script on the host and added some AI-related libs and includes to the existing sysroot by downloading the Vitis AI dependencies **vitis_ai_2021.1-r1.4.0.tar.gz** and extract it to the sysroot as the tutorial was guiding me to do.

Should I pull the Vitis AI with version 1.4 instead? If yes, which of the versions in this [link](https://hub.docker.com/r/xilinx/vitis-ai/tags) should I pull? The command **docker pull xilinx/vitis-ai:1.4** does not work.

KrishnaGaihre, 2024-03-22:
You can clone/follow Vitis AI 1.4 from [github ](https://github.com/Xilinx/Vitis-AI/tree/1.4)and docker pull 1.4.916 from [docker hub](https://hub.docker.com/r/xilinx/vitis-ai/tags) [docker pull xilinx/vitis-ai:1.4.916].
And yes you have to match the Vitis AI runtime version in Board (see it by running **xdputil query** on board's terminal) and the GPU/CPU docker for compiling the model.

closed 2024-04-02 by thallesmm