Xilinx/Vitis-AI #1424: PetaLinux boot stuck after chang the device tree of dpu

Yuxin-Yu, 2024-04-11:
I used the Vivado flow in DPUCZDX8G_VAI_v3.0 to run DPU. During the process of building the PetaLinux image, I used the device tree information generated by Vitis (obtained from the xsa file) to modify the project-spec/meta-user/recipes-bsp/device-tree/files/system-user.dtsi as follows:
```
/include/ "system-conf.dtsi"
/ {
};
&amba{
    hier_dpu_DPUCZDX8G: dpuczdx8g@8f000000 {
        /* This is a place holder node for a custom IP, user may need to update the entries */
        clock-names = "s_axi_aclk", "dpu_2x_clk", "dpu1_2x_clk", "m_axi_dpu_aclk";
        clocks = <&zynqmp_clk 71>, <&misc_clk_0>, <&misc_clk_0>, <&misc_clk_1>;
        compatible = "xlnx,dpuczdx8g-4.1";
        interrupt-names = "dpu0_interrupt", "dpu1_interrupt", "sfm_interrupt";
        interrupt-parent = <&gic>;
        interrupts = <0 89 4 0 90 4 0 91 4>;
        reg = <0x0 0x8f000000 0x0 0x1000000>;
    };
    misc_clk_0: misc_clk_0 {
        #clock-cells = <0>;
        clock-frequency = <249975000>;
        compatible = "fixed-clock";
    };
    misc_clk_1: misc_clk_1 {
        #clock-cells = <0>;
        clock-frequency = <124987500>;
        compatible = "fixed-clock";
    };
};
```
The modified PetaLinux startup is stuck (it can start PetaLinux normally before modification), and the stuck information is as follows:
```
U-Boot 2022.01 (Sep 20 2022 - 06:35:33 +0000)

CPU:   ZynqMP
Silicon: v3
Model: ZynqMP ZCU106 RevA
Board: Xilinx ZynqMP
DRAM:  4 GiB
PMUFW:  v1.1
PMUFW no permission to change config object
EL Level:       EL2
Chip ID:        zu7e
NAND:  0 MiB
MMC:   mmc@ff170000: 0
Loading Environment from FAT... *** Error - No Valid Environment Area found
*** Warning - bad env area, using default environment

In:    serial
Out:   serial
Err:   serial
Bootmode: LVL_SHFT_SD_MODE1
Reset reason:   EXTERNAL
Net:   FEC: can't find phy-handle
...
    3.060046] usbcore: registered new interface driver usbhid
[    3.065448] usbhid: USB HID core driver
[    3.072153] ARM CCI_400_r1 PMU driver probed
[    3.072769] fpga_manager fpga0: Xilinx ZynqMP FPGA Manager registered
[    3.080305] usbcore: registered new interface driver snd-usb-audio
[    3.086838] pktgen: Packet Generator for packet performance testing. Version: 2.75
[    3.094266] Initializing XFRM netlink socket
[    3.097868] NET: Registered PF_INET6 protocol family
[    3.103145] Segment Routing with IPv6
[    3.106384] In-situ OAM (IOAM) with IPv6
[    3.110317] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    3.116449] NET: Registered PF_PACKET protocol family
[    3.121167] NET: Registered PF_KEY protocol family
[    3.125930] can: controller area network core
[    3.130265] NET: Registered PF_CAN protocol family
[    3.134999] can: raw protocol
[    3.137941] can: broadcast manager protocol
[    3.142095] can: netlink gateway - max_hops=1
[    3.146483] Bluetooth: RFCOMM TTY layer initialized
[    3.151269] Bluetooth: RFCOMM socket layer initialized
[    3.156376] Bluetooth: RFCOMM ver 1.11
[    3.160085] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.165363] Bluetooth: BNEP filters: protocol multicast
[    3.170556] Bluetooth: BNEP socket layer initialized
[    3.175487] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    3.181369] Bluetooth: HIDP socket layer initialized
[    3.186323] 8021q: 802.1Q VLAN Support v1.8
[    3.190552] 9pnet: Installing 9P2000 support
[    3.194707] Key type dns_resolver registered
[    3.199058] registered taskstats version 1
[    3.202997] Loading compiled-in X.509 certificates
[    3.208782] Btrfs loaded, crc32c=crc32c-generic, zoned=no, fsverity=no
[    3.222818] ff000000.serial: ttyPS0 at MMIO 0xff000000 (irq = 62, base_baud = 6249375) is a xuartps
[    3.231850] printk: console [ttyPS0] enabled
[    3.231850] printk: console [ttyPS0] enabled
[    3.236144] printk: bootconsole [cdns0] disabled
[    3.236144] printk: bootconsole [cdns0] disabled
[    3.245679] ff010000.serial: ttyPS1 at MMIO 0xff010000 (irq = 63, base_baud = 6249375) is a xuartps
[    3.258874] of-fpga-region fpga-full: FPGA Region probed
[    3.266395] xilinx-zynqmp-dma fd500000.dma-controller: ZynqMP DMA driver Probe success
[    3.274469] xilinx-zynqmp-dma fd510000.dma-controller: ZynqMP DMA driver Probe success
[    3.282542] xilinx-zynqmp-dma fd520000.dma-controller: ZynqMP DMA driver Probe success
[    3.290611] xilinx-zynqmp-dma fd530000.dma-controller: ZynqMP DMA driver Probe success
[    3.298675] xilinx-zynqmp-dma fd540000.dma-controller: ZynqMP DMA driver Probe success
[    3.306742] xilinx-zynqmp-dma fd550000.dma-controller: ZynqMP DMA driver Probe success
[    3.314812] xilinx-zynqmp-dma fd560000.dma-controller: ZynqMP DMA driver Probe success
[    3.322872] xilinx-zynqmp-dma fd570000.dma-controller: ZynqMP DMA driver Probe success
[    3.331009] xilinx-zynqmp-dma ffa80000.dma-controller: ZynqMP DMA driver Probe success
[    3.339078] xilinx-zynqmp-dma ffa90000.dma-controller: ZynqMP DMA driver Probe success
[    3.347150] xilinx-zynqmp-dma ffaa0000.dma-controller: ZynqMP DMA driver Probe success
[    3.355218] xilinx-zynqmp-dma ffab0000.dma-controller: ZynqMP DMA driver Probe success
[    3.363287] xilinx-zynqmp-dma ffac0000.dma-controller: ZynqMP DMA driver Probe success
[    3.371350] xilinx-zynqmp-dma ffad0000.dma-controller: ZynqMP DMA driver Probe success
[    3.379416] xilinx-zynqmp-dma ffae0000.dma-controller: ZynqMP DMA driver Probe success
[    3.387481] xilinx-zynqmp-dma ffaf0000.dma-controller: ZynqMP DMA driver Probe success
[    3.395767] xilinx-zynqmp-dpdma fd4c0000.dma-controller: Xilinx DPDMA engine is probed

```
Does anyone know how to solve it?

Yuxin-Yu, 2024-04-19:
Closing FPGA Manage in the petalinux configuration can solve the problem while ensuring the correctness of the device tree

closed 2024-04-19 by Yuxin-Yu

mjayaths, 2024-04-24:
Hi,  I'm having the same issue. Petalinux ver 2022.2 DPU IP 4.1 VAI 3.0.
When DPU is added to the design, Petalinux boot up stops at the above point. FPGA manager is not selected in the petalinux configuration. 
Any ideas why this issue happens ?
Thanks