Xilinx/Vitis-AI #1431: Example running failed in KR260

hurricanemad, 2024-05-02:
When I try the examples of Vitis-AI master branch in the KR260, the error shown by the following figure bump out.
Actually, I refer to the instruction of https://xilinx.github.io/Vitis-AI/3.0/html/docs/quickstart/mpsoc.html. However, error occurs.

F0502 12:17:32.779125 16674 dpu_runner.cpp:211] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot open library! lib=libvart-dpu-runner.so, error=ERROR CODE: /lib/libvart-xrt-device-handle.so.3: undefined symbol: _ZN6google15ErrnoLogMessageC1EPKciimMNS_10LogMessageEFvvE
*** Check failure stack trace: ***
Aborted (core dumped)

[image]

warhammercasey, 2024-09-03:
Same issue here. Did you ever find a fix?

quentonh, 2024-09-04:
@hurricanemad @warhammercasey This issue states the target as the KR260 and not the KV260.  Though it is the same SOM, I have no verification that anyone has ever specifically tested the pre-built image on the KR260 Starter Kit.  It may function as-is, but I do not know this to be the case.  Or, are you specifically building an image for the KR260 and hitting this issue?

warhammercasey, 2024-09-04:
> @hurricanemad @warhammercasey This issue states the target as the KR260 and not the KV260. Though it is the same SOM, I have no verification that anyone has ever specifically tested the pre-built [...]

It turns out the issue was downloading the prebuilt vart .tar.gz file in the guide @hurricanemad linked. I was able to fix it by cloning and building vart instead of using that package.

quentonh, 2024-09-04:
So, for confirmation @warhammercasey this is indeed on the KR260 (not KV260)? Great work finding the solution.

hurricanemad, 2024-09-06:
The error occurs when I run the sample code of Vitis-AI on KR260. When I first run the C++ example, the error will pop up. This would be a bug on KR260 SOM. So you have to run the python example code. And the fan on KR260 SOM rotate fiercely. This means the DPU kernel have run. Then, you try to run the C++ example. This problem would be solved. This method could take effect.

warhammercasey, 2024-09-06:
> So, for confirmation @warhammercasey this is indeed on the KR260 (not KV260)? Great work finding the solution.

My issue was actually on a ZUBoard 1CG. So entirely user error. I didnt realize the packages in that download link were specifically for that board image

mj1028, 2024-11-13:
I’m experiencing a similar issue with the Vitis AI library on a KV260 board. Running ./test_jpeg_classification with resnet50_pt gives the error: symbol lookup error: /usr/lib/../lib/../lib/libvart-xrt-device-handle.so.3: undefined symbol: xclClose. Has anyone found a stable solution? Should I rebuild the VART library from source, or are there specific versions of XRT and Vitis AI recommended to resolve this? Any guidance would be greatly appreciated!