Xilinx/Vitis-AI #1436: dpuczdx8g v4.1 compatibility consultation

Yuxin-Yu, 2024-05-23:
Hi， I would like to ask if the latest dpuczdx8g v4.1 version can run on zynq 7000 series devices by disabling the softmax module?

quentonh, 2024-05-23:
@Yuxin-Yu we have done limited testing, but yes, we can confirm that the DPUCZ remains compatible with Vitis AI with a minor patch to the Vitis AI Library.
[vitis-ai-library.zip](https://github.com/Xilinx/Vitis-AI/files/15419306/vitis-ai-library.zip)

Yuxin-Yu, 2024-05-24:
Thank you @quentonh . In addition, in order to make dpuczdx8g v4.1 compatible with zynq 7000, I have changed some related parameters, as shown in the red circles in the following two figures. Is that right?
[image]
[image]

Yuxin-Yu, 2024-05-24:
Also, it should be noted that I am using Vitis AI 3.0 version. Are the patch files used the same? @quentonh

quentonh, 2024-05-27:
@Yuxin-Yu we have not internally verified the patches against Vitis AI 3.0, but the modifications to the Vitis AI Library will be very similar if not identical.  The required changes result from differences in the NEON instruction set between the Cortex A53 (ZU+) and Cortex A9 (Zynq-7000).  You should do compare between the modified source files in the two releases (3.5 and 3.0) to ensure that the changes apply.

With regard to integration into a hardware platform, I can send you an example.  Please drop me an email to quentonh at amd.com.

Yuxin-Yu, 2024-05-28:
Hi @quentonh .I don't know how to send email at amd.com.If possible, you can send the example to my email : yyx15910878756@gmail.com.Thank you.

Yuxin-Yu, 2024-06-04:
Hello @quentonh , I successfully ran the resnet50 inference process using zynq7000, but the inference result is incorrect, as shown below:
```
root@debian:~/Vitis-AI/src/vai_runtime/vart_debug/softmax-runner-cpu/test/resnet_v1_50_softmax_cpu# ./run.sh ~/app-riscv/img/bellpeppe-994958.JPEG 

score[971]  =  0.301356     text: bubble,
score[616]  =  0.301356     text: knot,
score[506]  =  0.301356     text: coil, spiral, volute, whorl, helix,
score[488]  =  0.0247368    text: chain,
score[679]  =  0.0150036    text: necklace,
```

What could be the reason and how should I debug it?

Yuxin-Yu, 2024-06-21:
Problem solved, it's because my xmodel file was used incorrectly

closed 2024-06-21 by Yuxin-Yu