Xilinx/Vitis-AI #1450: Resnet-50 test accuracy and loss of accuracy in DPU

rattokiller, 2024-07-24:
Hi there,

I am trying to run the Resnet-50 adestrated on imagenet that you give as an [example](https://github.com/Xilinx/Vitis-AI/tree/3.0/examples/vai_runtime/resnet50), I am using vitis3.0.

The net is running in ZCU102.

To your example code I have made a few modifications for the calculation of top1 and top5.

I find that these values are 23% top1 and 42% top5. (test with 50 images for calass = 50000 images)

Do you agree with these values? 
What values does your network get after quantization?

Moreover, for various CASTOM networks (RESNET 8,18 etc) on CIFAR10, I find myself with a much lower accuracy between the quantized (which is usually lower than the floating-point model) (30-40 % accuracy).

By any chance can you tell me how much accuracy loss do you find between a network performed after quantization and on DPU?

Thank you.

quentonh, 2024-07-24:
@rattokiller the verified quantization accuracy of the models in the Vitis AI Model Zoo is published [here](https://xilinx.github.io/Vitis-AI/3.0/html/_downloads/08f25862a0dd85eb493ab84edb004a0e/ModelZoo_VAI3.0_Github.xlsx)

You should be able to achieve this level of accuracy if you leverage the model files that are provided in the Zoo.  Both the trained and quantized and source floating-point models for ResNet50 are provided.

rattokiller, 2024-07-25:
@quentonh Thank you for your reply.

With the quantized accuracy values measured by your conda that you have provided, I find myself.

In general I also find myself with the performance parameters.

Instead I am asking for the accuracy parameters calculated directly on the DPU, in the [file](https://view.officeapps.live.com/op/view.aspx?src=https%3A%2F%2Fxilinx.github.io%2FVitis-AI%2F3.0%2Fhtml%2F_downloads%2F08f25862a0dd85eb493ab84edb004a0e%2FModelZoo_VAI3.0_Github.xlsx&wdOrigin=BROWSELINK) you shared with me I only see the performance parameters per board.

Is the quantized accuracy the same as what you measure from the DPU?

quentonh, 2024-07-25:
@rattokiller The quantized and floating-point accuracy for these models are found in that spreadsheet as well.  The accuracy is validated for each model that is provided in the Model Zoo, and the results are published in columns L and M in that spreadsheet.

Does that help?

rattokiller, 2024-07-29:
@quentonh the quantized and floating-point accuracy is run in conteiner.

I request the accuracy validation with run model on DPU

quentonh, 2024-07-29:
@rattokiller I see.  If you are seeing discrepancies between the accuracy reported in the Docker Container and the accuracy in hardware, I would recommend that you request assistance from your local FAE.  They will help you to escalate the issue to our technical support team.

quentonh, 2024-07-29:
Also, @rattokiller please see:  https://docs.amd.com/r/en-US/ug1354-xilinx-ai-sdk/Model-Accuracy-Test