Xilinx/Vitis-AI #1452: When release 4.0 version?

wangguoqing129, 2024-07-27:
I am trying to deploy yolov8 in zcu104 with vitis ai 3.5, but it is not well supported. So if 4.0 version good for zcu104 and yolov8 model? When will you update 4.0 version?

Thank you for your reply.

jimheaton, 2024-07-29:
Hi @wangguoqing129, 
You can use the VIA 3.5 Library examples for yolov8, but you will need to rebuild the ZCU104 DPU for VAI3.5. 
This is not officially supported by AMD, but I do I have git repo to build the DPU_TRD that I'm attaching.
You will need Vivado 2023.2 and Petalinux 2023.2 to build this.

You will need a .prototxt file to run the via yolov8 examples, here's an example for yolov8m

model {
  kernel {
     mean: 0.0
     mean: 0.0
     mean: 0.0
     scale: 0.00392157
     scale: 0.00392157
     scale: 0.00392157
  }
  model_type : YOLOv8
  yolo_v8_param {
    stride:8
    stride:16
    stride:32
    max_boxes_num:30000
    max_nms_num:300
    num_classes: 80
    detect_layer_name: "n0_cat__ret_263_fix"
    detect_layer_name: "n1_cat__ret_278_fix"
    detect_layer_name: "n2_cat__ret_293_fix"
    conf_thresh: 0.3
    nms_threshold: 0.7
  }
}

[VAI-3.5-ZUP-DPU-TRD-main.zip](https://github.com/user-attachments/files/16415791/VAI-3.5-ZUP-DPU-TRD-main.zip)

quentonh, 2024-08-01:
@claganier @jimheaton will confirm (I believe that he attempted this, but I am uncertain of the answer).  However, I am still concerned that the activation type will be a bit of a blocker for you (per my email).  @jimheaton have you ever attempted to deploy YOLOv8 on the DPUCZ?

jimheaton, 2024-08-02:
Hi @claganier, You should be able to use a similar method to convert the KV260 to VAI 3.5.
What you would need to do is take the KV260 bsp, uncompress it, and modify the  contained Vivado project to add the DPU and generate the xsa. This is what the first step in the repo was doing for the ZCU104 and 102 boards.

Once you have the .xsa, you can build Linux using the flow in the repo, with the only change would be using the KV260 bsp.

The way the KV260 boots is different than the ZC104 that the repo was designed for, so their may be some changes needed in how to wic image is created.

When I have some time, I can try porting the KV260 to VAI 3.5, but it  might be few weeks.

quentonh, 2024-08-02:
Thanks @jimheaton!  @claganier also please note that you will need Jim's Vitis AI Library patch that can be found here:  https://github.com/Xilinx/Vitis-AI/files/15419306/vitis-ai-library.zip

jimheaton, 2024-08-02:
Hi @quentonh, Yes I believe that I did get this to run on the ZCU102 using the vai-library sw examples after migrating to VAI 3.5.

quentonh, 2024-08-02:
Posting my "thoughts" on YOLOv8 deployment on the DPUCZ, FWIW:

~~~~~~~~~~~~~~~~~~~~~~

Having thought about this a little more, I realize now that the DPUCZ supports Hard Swish.  You may be able to use Hard Swish in place of Hard SiLU.  You can see [here](https://pytorch.org/docs/stable/generated/torch.nn.SiLU.html) that the SiLU activation function is also known as the Swish function.   Note the curve of the transfer function.  I believe that Hard SiLU is simply an approximation of SiLU, but for the moment I cannot find the transfer function for Hard SiLU.  

In 2019, the MobileNetv3 paper introduced a new non-linear activation type called h-Swish, or Hard Swish, or hard_swish that was intended to be easier to compute and easier to quantize.  Why does this matter?  It matters because Hard Swish is in fact a piecewise approximation of SiLU:  [Hardswish — PyTorch 2.4 documentation](https://pytorch.org/docs/stable/generated/torch.nn.Hardswish.html)  Note the similarity between the Hard Swish and SiLU transfer functions.  They are not identical, but overall, it is a reasonable approximation of SiLU.

Now, if you quickly look at the computations for tf.keras.activations.hard_silu  |  TensorFlow v2.16.1, you will see that in fact it is nearly identical to the hard_silu function.

So, in fact, HardSiLu and HardSwish, SiLU and SiLU are “approximately” the same.

You can thus try to swap the HardSiLU activation type in YOLOv8 with HardSwish activation type, re-train your network and attempt to deploy it on the DPUCZ.  What I cannot promise you is that the Hard Swish activation type will be supported by the quantizer and compiler IF the preceding and succeeding layer types found in YOLOv8 do not appropriately merge with that activation.  This may require more extensive modification than simply swapping out the activation in the network.  Leveraging the Vitis AI 3.5 (not 3.0) Model Inspector may help you better evaluate the compatibility of the network before attempting to optimize accuracy through extensive re-training.

Net result, this will likely require some effort, but I believe it may be a potential option.  Note that when activations are swapped, there is often some loss of accuracy.  In some cases, the user can modify the network further (potentially increasing the depth of the network or increasing the input resolution) to compensate.

quentonh, 2024-08-05:
@claganier Sorry!  I made a dumb mistake.  The patch is only required for Zynq-7000, and not for Zynq Ultrascale+.

claganier, 2024-08-28:
Hi @jimheaton, thank you for these instructions to setup VAI 3.5 on the Kria KV260. 

I have setup Vivado on my machine and have tried going about the modification but am facing a few issues: 

1. I downloaded the `xilinx-kv260-starterkit-2024.1.bsp` from https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html, is this the correct BSP file required?
2. I have tried to open the hardware project file in Vivado (2023.2 as mentioned in your previous message) but the KV260 project seems to have been created with a more recent version of Vivado. Should I upgrade to a more recent Vivado version or should I use another BSP ?
3. Would you be able to provide instructions regarding how to add the DPU and generate the XSA? I do not have experience with Vivado and am struggling to see how I would go about that.

Thank you

jimheaton, 2024-09-05:
Hi @claganier, 
I  used 2023.2 so you need to download the 2023.2 bsp.
Once you have the design opened in Vivado, you would want to connect the DPU similar to whats shown in the DPU Vivado TRD: https://github.com/user-attachments/files/16415791/VAI-3.5-ZUP-DPU-TRD-main.zip

claganier, 2024-09-09:
Hello @jimheaton, thank you for getting back to me. I have tried using the repo you provided, and did the following modification: 
1.	Modified `main.tcl` to accept KV260 as board name – setting the project board to “xilinx.com:kv260_som:part0:1.4”
2.	Added a `kv260.xdc` file based on the `default.xdc` from `xilinx-kv260-starterkit-2023.2`:
```
# Copyright (C) 2022, Xilinx, Inc. 
# Copyright (C) 2022, Advanced Micro Devices, Inc.
# SPDX-License-Identifier: Apache-2.0

set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]

#Fan Speed Enable
set_property PACKAGE_PIN A12 [get_ports {fan_en_b}]
set_property IOSTANDARD LVCMOS33 [get_ports {fan_en_b}]
set_property SLEW SLOW [get_ports {fan_en_b}]
set_property DRIVE 4 [get_ports {fan_en_b}]
```
3.	Created a blank `dpu_kv260.tcl` file

With these modifications I am getting the following error when running the `vivado -mode batch -source main.tcl -tclargs kv260`:
```
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/colin/Downloads/VAI-3.5-ZUP-DPU-TRD-main/hw/dpu_ip'.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2023.2/data/ip'.
# source dpu_${board}.tcl
# save_bd_design
ERROR: [BD 5-104] A block design must be open to run this command. Please create/open a block design.
ERROR: [Common 17-39] 'save_bd_design' failed due to earlier errors.
    while executing "save_bd_design"   (file "main.tcl" line 25)
```
I am not sure what is the error - if I was meant to have `dpu_kv260.tcl` filled up with material (if so not sure what would go in there as `zcu102` and `zcu104` are different), or if another error occurred. 

Thank you very much for you help.