Xilinx/Vitis-AI #1455: Vck 190 setup problems.

meltedhyperion, 2024-08-11:
## So here's the breakdown!

1) we have been using the same installation setup provided in the documentation.( https://xilinx.github.io/Vitis-AI/3.0/html/docs/quickstart/vck190.html )

2) We are using Ubuntu OS on host machine.

3) All the connections have been done as per the guide.

4) We are using picocom for serial communication with the board from the host machine

## The Problem

1) In Ubuntu we are shown the connection as ttyUSB0, ttyUSB1, ttyUSB2 and ttyUSB3, we try emulating the UART connection to all of them using picocom but only USB3 connects to the boot process of the System controller linux terminal.

2) In the quick start guide it mentions, the boot process should start with the process "Xilinx Versal Platform Loader and Manager", but its the starting "Xilinx zynq MP first stage boot loader."

3) It mentions that the board already includes vitis ai libraries and required runtime, but it didnt. I was trying to run a model of xmodel format by following the same guide but it some of the directories and files mentioned did not exist (i was running it in the first stage boot os terminal).
```
[Target] $ cd ~
[Target] $ tar -xzvf vitis_ai_runtime_r3.0.0_image_video.tar.gz -C Vitis-AI/examples/vai_runtime
```
Vitis-AI folder did not exist, so i cloned from from this repo and copied it to the board using scp.

```
[Target] $ ./resnet50 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel
```
./resnet50 is a directory and not an executable file.

can someone help me out. I think the main issue is that I am not able to get the "Xilinx Versal Platform Loader and Manager" terminal to connect.

@hanxue @eemars @quentonh @woinck @mahinlma

quentonh, 2024-08-12:
@meltedhyperion The System Controller is a Zynq Ultrascale+ device on the VCK190 that does system and board level management.  In other words, I believe you are trying to install the libraries and run the model on the System Controller ZU+ device.

The critical first step needs to be to establish a connection to UART0 (the first of the three emulated COM ports).  UART0 communicates with the Versal target.

A few things to check:

[1] Did the System Controller boot successfully?  Yes.
[2] Can you boot from QSPI and connect to UART0 when booting from QSPI?
[3] Is the SD Card for the Versal image faulty?
[4] Is there a problem with the VCK190 Vitis AI SD Card image file that was downloaded?  Can you boot the board from an alternative VCK190 image such as this one:  https://xilinx.github.io/vck190-base-trd/2022.1/html/run/run-setup.html

meltedhyperion, 2024-08-13:
Yes that's what I was doing apparently, and that's the wrong approach, right?!
I tried connecting to UART0. I am using ubuntu on my host machine and it shows `4 Serial connections` for VCK190 (ttyUSB0, ttyUSB1, ttyUSB2 and ttyUSB3 (where the System controller terminal is opening on USB3))
I tried all 4 and only USB3 responds by opening the emulated terminal for system controller (i am using picocom).

Do i need to flash an image on the Versal SD card as well? The SysCtrl SD card image comes with a pre flashed image, and the Versal one is empty

MeltedHyperion07, 2024-08-30:
@quentonh what should be done here?

> @quentonh Yes that's what I was doing apparently, and that's the wrong approach, right?! I tried connecting to UART0. I am using ubuntu on my host machine and it shows `4 Serial connections` for VCK190 (ttyUSB0, ttyUSB1, ttyUSB2 and ttyUSB3 (where the System controller terminal is opening on USB3)) I tried all 4 and only USB3 responds by opening the emulated terminal for system controller (i am using picocom).
> 
> Do i need to flash an image on the Versal SD card as well? The SysCtrl SD card image comes with a pre flashed image, and the Versal one is empty

quentonh, 2024-09-04:
@meltedhyperion Sorry for the delayed response.  I was OOO for the past two weeks.  Yes, you need to ensure that a boot image is available for the Versal SoC before the Versal serial ports will be active.  In the context of the Quickstart tutorial, instructions are provided to obtain the VCK190 Versal SD card image.  You should download and write that image to a suitable SD card for the Versal SoC.  The System Controller image should be similarly loaded on the SC SD card.  You cannot mix these up....nothing will work.  Also, you have to configure the mode switches for SD boot for the Versal SoC.  I believe that all of this is reasonably well called out in the Quickstart tutorial, but perhaps something was not clear?

It sounds almost as though you may benefit by using the VCK190 getting started Wiki to cross-check that everything is working as expected:  https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/748617729/Versal+AI+Core+Series+VCK190+Evaluation+Kit

MeltedHyperion07, 2024-10-16:
@quentonh you can close this issue now. It is now resolved. It was an issue with the Versal Image. I resolved it by reflashing the original image.

closed 2024-12-18 by meltedhyperion