Xilinx/Vitis-AI #1460: PermissionError: [Errno 13] Permission denied: 'quantize_result'

MeltedHyperion07, 2024-09-06:
Hey, I am learning how to use Vitis AI 3.0 and trying to run the Quickstart tutorial for Vitis AI 3.0 `VCK190` resnet18.

At the Section of the "Pytorch turorial" :

 
`
Step 7 : Next, let’s run the Model Inspector to confirm that this model should be compatible with the target DPU architecture.`:

I am getting this error

 
```
(vitis-ai-pytorch) vitis-ai-user@HILLS:/workspace/resnet18$ sudo python resnet18_quant.py --quant_mode float --inspect --target DPUCVDX8G_ISA3_C32B6 --model_dir model
Traceback (most recent call last):
  File "/workspace/resnet18/resnet18_quant.py", line 8, in <module>
    from pytorch_nndct.apis import torch_quantizer
ModuleNotFoundError: No module named 'pytorch_nndct'
(vitis-ai-pytorch) vitis-ai-user@HILLS:/workspace/resnet18$ python resnet18_quant.py --quant_mode float --inspect --target DPUCVDX8G_ISA3_C32B6 --model_dir model

[VAIQ_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test 

[VAIQ_NOTE]: Inspector is on.

[VAIQ_NOTE]: =>Start to inspect model...
Traceback (most recent call last):
  File "resnet18_quant.py", line 314, in <module>
    file_path=file_path)
  File "resnet18_quant.py", line 234, in quantization
    inspector.inspect(quant_model, (input,), device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 196, in inspect
    self._inspector_impl.inspect(module, input_args, device, output_dir, verbose_level)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/hardware_v3/inspector.py", line 52, in inspect
    dev_graph, deploy_graphs = prepare_deployable_graph(copied_model, input_args, device, output_dir)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/hardware_v3/utils.py", line 30, in prepare_deployable_graph
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/pytorch_nndct/qproc/utils.py", line 215, in prepare_quantizable_module
    nndct_utils.create_work_dir(export_folder)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/site-packages/nndct_shared/utils/io.py", line 144, in create_work_dir
    os.makedirs(dir_name)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.7/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: 'quantize_result'
```

Can someone help me out here with this issue.

@quentonh @hanxue @eemars @woinck @mahinlma

amhdvi98, 2024-11-04:
i am also getting the same error when running vitis ai optimizer with one step pruning.

cjwright96, 2024-12-03:
I'm getting the second error - the first is because using sudo changes the python env, but the permissions error I am stuck on as well.