Xilinx/Vitis-AI #1461: Cnt donwload .zip

lida1125, 2024-09-08:
I want to Download and unzip [mpsoc common system](https://www.xilinx.com/member/forms/download/xef.html?filename=xilinx-zynqmp-common-v2022.1.tar.gz) & [zcu102 base platform](https://www.xilinx.com/member/forms/download/design-license-zcu102-base.html?filename=xilinx_zcu102_base_202210_1.zip) package.

But Unable to find archive with filename: xilinx_zcu102_base_202210_1.zip code.

How should I run this example?thank you!
(Vitis-AI/examples/waa/apps/adas_detection/build_flow/DPUCZDX8G_zcu102)

quentonh, 2024-09-09:
@lida1125 Such packages are available from www.xilinx.com/download, but to be more specific with respect to your request, please download the package from the following URL:  https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-platforms/archive-vitis-embedded.html

BitaAsghari, 2024-09-22:
I'm facing the same issue. I need to implement YOLO on my ZCU102 board, but the files mentioned in this example are missing.
Example link: 
https://github.com/Xilinx/Vitis-AI/blob/master/examples/waa/apps/adas_detection/pre_built_flow/DPUCZDX8G_zcu102/README.md