Xilinx/Vitis-AI #1467: ERROR: [VPL 60-1328] Vpl run 'vpl' failed

a42385029, 2024-09-20:
in Vitis can't build project:
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
WARNING: [VPL 60-1142] Unable to read data from '/home/zhr/Project/dpu_vitis1/alinx_9eg_system_hw_link/Hardware/dpu.build/link/vivado/vpl/output/generated_reports.log', generated reports will not be copied.
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [16:09:25] Run run_link: Step vpl: Failed
Time (s): cpu = 00:01:07 ; elapsed = 00:01:49 . Memory (MB): peak = 458.723 ; gain = 0.000 ; free physical = 1036 ; free virtual = 1411
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [makefile:58: dpu.xclbin] Error 1

Yanze66, 2024-11-15:
I had the same issue. Sad

a42385029, 2024-11-15:
> I had the same issue. Sad
I have solved this problem

a42385029, 2024-11-15:
> I had the same issue. Sad

[image]
Vivado->Tool->settin->Implementation,Disable

a42385029, 2024-11-15:
There should be no .dcp file in the .xsa file