Xilinx/Vitis-AI #1488: vitis ai xmodel not generated in vitis AI 3.5

ChristiaanBoe, 2024-11-01:
Hello everyone,

When generating a bitstream for my CNN model I seem to run into the problem of no model being generated using vitis AI 3.5, however I do not seem to get any error message. Could anyone give me a hint what I did wrong? I have attached my code below:

```
import os
import sys
import argparse
import torch
import numpy as np
import scipy
import scipy.signal
import random
from tqdm import tqdm
from pytorch_nndct.apis import torch_quantizer

device = torch.device("cuda" if torch.cuda.is_available() else "cpu")

parser = argparse.ArgumentParser()

parser.add_argument(
    '--model_dir',
    default="./",
    help='Trained model file path.')
parser.add_argument(
    '--config_file',
    default=None,
    help='Quantization configuration file')
parser.add_argument(
    '--batch_size',
    default=32,
    type=int,
    help='Input data batch size')
parser.add_argument('--quant_mode', 
    default='calib', 
    choices=['float', 'calib', 'test'], 
    help='Quantization mode. float: evaluate float model, calib: calibrate for quantization, test: evaluate quantized model')
parser.add_argument('--fast_finetune', 
    dest='fast_finetune',
    action='store_true',
    help='Fast finetune model before calibration')
parser.add_argument('--deploy', 
    dest='deploy',
    action='store_true',
    help='Export xmodel for deployment')

parser.add_argument('--target', 
    dest='target',
    nargs="?",
    const="",
    help='specify target device')

args, _ = parser.parse_known_args()

def generate_win_coeffs(M, P, window_fn="hamming"):
    win_coeffs = scipy.signal.get_window(window_fn, M*P)
    sinc       = scipy.signal.firwin(M * P, cutoff=1.0/P, window="rectangular")
    win_coeffs *= sinc
    return win_coeffs

class FFTLayer(torch.nn.Module):
    def __init__(self, input_size):
        super(FFTLayer, self).__init__()
        self.input_size = input_size
        self.FFTconv = torch.nn.Conv2d(input_size, input_size * 2, kernel_size=(1, 1), bias=False)

        F = torch.from_numpy(np.fft.fft(np.eye(self.input_size)))
        self.FFTconv.weight.data[0:self.input_size, :, :, :] = torch.unsqueeze(torch.unsqueeze(F.real.float(), -1), -1)
        self.FFTconv.weight.data[self.input_size:(self.input_size * 2), :, :, :] = torch.unsqueeze(torch.unsqueeze(F.imag.float(), -1), -1)

        #self.FFTconv.weight.data.requires_grad = False

    def forward(self, x):
        output = self.FFTconv(x)
        return output

class PFB_FIR_FFT(torch.nn.Module):
    def __init__(self, win_coeffs, M, P, expected_input_size):
        super(PFB_FIR_FFT, self).__init__()
        self.win_coeffs = win_coeffs.reshape((M, P)).T
        self.win_coeffs = self.win_coeffs.unsqueeze(0).unsqueeze(1)
        self.win_coeffs = self.win_coeffs.view(P, 1, 1, M)
        self.P = P
        self.M = M
        self.size = expected_input_size
        self.W = int(self.size / self.M / self.P)
        self.Maxsize = self.M * self.W - self.M
        self.WM = self.M * self.W

        self.FIR = torch.nn.Conv2d(in_channels=self.P, out_channels=self.P, kernel_size=(1, self.M), stride=(1, 1), padding=(0, 0), bias=False, groups=self.P)
        self.FFTlayer = FFTLayer(input_size=self.P)
        self.FIR.weight = torch.nn.Parameter(self.win_coeffs)

    def forward(self, input):
        input = self.FIR(input)
        input = self.FFTlayer(input)
        return input

def load_data(M, P, batch_size):
    # Generate random data
    x = np.sin(np.arange(0, M * P * 16) / np.pi)
    x = np.random.rand(*x.shape).astype(np.float32)
    x = np.expand_dims(x, axis=0)
    x = np.expand_dims(x, axis=0)
    xinput = np.random.rand(batch_size,*x.shape)
    W = int(xinput.shape[-1] / M / P)
    WM = M * W
    xinput = torch.from_numpy(xinput).float()
    xinput = xinput.view(xinput.shape[0], WM, 1, P).permute(0, 3, 2, 1)[:, :, :, 0: WM-1]
    return xinput

def evaluate(model, input):
    model.eval()
    input = input.to(device)
    output = model(input)
    return output

def quantization(model, M, P, target):
    batch_size = args.batch_size
    quant_mode = args.quant_mode
    config_file = args.config_file
    deploy = args.deploy

    input = load_data(M, P, batch_size)
    #W = int(input.shape[-1] / M / P)
    #WM = M * W
    #input = input.view(batch_size, M * P, 1, -1).to(device)
    #input = input.view(input.shape[0], WM, 1, P).permute(0, 3, 2, 1)[:, :, :, 0: WM-1]

    if quant_mode == 'float':
        quant_model = model
    else:
        quantizer = torch_quantizer(
            quant_mode, model, (input), device=device, quant_config_file=config_file, target=target)
        quant_model = quantizer.quant_model

    output = evaluate(quant_model, input)
    print("Output Shape:", output.shape)

    if quant_mode == 'calib':
        quantizer.export_quant_config()
        print('running calib')
    if deploy:
        print('Deploying')
        #quantizer.export_xmodel(deploy_check=False, output_dir='.')
        #quantizer.export_torch_script()
        #quantizer.export_onnx_model()

        var = quantizer.export_xmodel(deploy_check=True)
        #import pdb; pdb.set_trace()
        print(var)

def main():
    M = 32
    P = 32
    model_dir = args.model_dir
    model_file = os.path.join(model_dir, "PFB_FIR_FFT.pth")

    # Generate window coefficients
    win_coeffs = generate_win_coeffs(M, P)
    win_coeffs = torch.from_numpy(win_coeffs).float()

    # Initialize the model
    model = PFB_FIR_FFT(win_coeffs=win_coeffs, M=M, P=P, expected_input_size=M * P * 16)
    model = model.to(device)

    # Quantize the model
    quantization(model, M, P, target=args.target)

if __name__ == '__main__':
    main()
```

Best regards,

Christiaan Boerkamp

jimheaton, 2024-11-01:
Please try again with deploy_check set to False:
var = quantizer.export_xmodel(deploy_check=False)

sandyagur8, 2024-11-05:
Hey there 
There are few things you can try.
1) You can only export xmodel if ```quant_mode == 'test'``` and ```batch_size==1```
2) Try doing a  forward pass with sample input before exporting the model.
3) Try with ```deploy_check=True```

Is there any specific error that you are seeing ? or its just like quantization completes successfully 
and no xmodel is being generated?