Xilinx/Vitis-AI #1489: tf_superpoint quantization scripts

CroosJJSE, 2024-11-01:
hi, i am an 4th year undergraduate, involved in an project where i am implementing customized superpoint (pytorch) based SLAM on KRIA KR260, 
I am struggling to do quantizing and pruning using Vitis AI,
I saw the tf_superpoint model on model zoo,
if i can get that quantizing and pruning python scripts of tf_superpoint, it will be helpful for my project,
Your help is so appreciable
Thank you

jimheaton, 2024-11-01:
I don't think this model was pruned, so we don't have scripts for that.
The training and quantization code should be here: https://www.xilinx.com/bin/public/openDownload?filename=tf_superpoint_3.5.zip

CroosJJSE, 2024-11-02:
thank you,
i will check it

closed 2024-11-02 by CroosJJSE