Xilinx/Vitis-AI #1491: Investigating inference errors on ZCU104: verifying the correctness of .xmodel judgments in Vitis AI

lroy615, 2024-11-06:
During the quantization process, I used the following commands: python quantize.py --quant_mode calib --subset_len 1 and python quantize.py --quant_mode test --subset_len 1 --batch_size=1 --deploy to generate DetectMultiBackend_int.xmodel. Then, I converted it into an .xmodel file executable on the DPU, integrating hardware information. However, when deploying the .xmodel model on the ZCU104 development board, numerous errors occurred, making it difficult to determine whether the issue lies with the inference script or the .xmodel itself. I would like to ask if anyone can confirm that the quantization steps were error-free and that the .xmodel file was generated correctly. I have attached a screenshot of the quantization results below.（PS：The version of Vitis-AI is 2.5.）
[image]
[image]
[image]
[image]
[image]

9temp, 2024-12-02:
能不能上传一下量化的代码

lroy615, 2024-12-04:
你好，这是我参考的github：https://github.com/kevinsu20/Vitis-ai-zcu104-yolov5.git
，里面有我参考的量化代码