Xilinx/Vitis-AI #1506: Can WAA speed up video reading/decoding?

mtntruong, 2025-02-14:
Hello,
I wrote an application for DPU inference, in which I re-used the `DecodeThread` from [demo.hpp](https://github.com/Xilinx/Vitis-AI/blob/1eed93cde2ff3ed841ddb03dff0714b4e2d34f5b/src/vai_library/benchmark/include/vitis/ai/demo.hpp). However, the speed is always slower than 18fps, regardless of network. I measured the running times of the components and found that reading/decoding video frames is the most time-consuming step. Specifically, I measured the time for reading a video frame as follows
```cpp
struct DecodeThread : public MyThread {
    ...
    virtual ~DecodeThread() {}

    virtual int run() override {
        auto& cap = *video_stream_.get();
        cv::Mat image;
        
        auto start = std::chrono::high_resolution_clock::now();
        cap >> image; // <-- timing for this step
        auto end = std::chrono::high_resolution_clock::now();
        std::chrono::duration<double> duration = end - start;
        std::cout << "Time taken: " << duration.count() << " seconds" << std::endl;

        auto video_ended = image.empty();
    ...
}
```
The printed outputs are
```
Time taken: 0.0663732 seconds
Time taken: 0.0521586 seconds
Time taken: 0.0611419 seconds
Time taken: 0.0549036 seconds
Time taken: 0.0649557 seconds
Time taken: 0.0468149 seconds
Time taken: 0.0560036 seconds
Time taken: 0.055309 seconds
Time taken: 0.0608081 seconds
Time taken: 0.0486924 seconds
Time taken: 0.0681339 seconds
Time taken: 0.0506587 seconds
Time taken: 0.060107 seconds

```
Reading video frames takes approximately 0.05 seconds, that means the inference speed cannot be higher than 20fps. I have read the [WAA examples](https://github.com/Xilinx/Vitis-AI/tree/1eed93cde2ff3ed841ddb03dff0714b4e2d34f5b/examples/waa), but there aren't any examples for accelerating video decoding. Can WAA improve video reading time? If yes, may I ask are there any tutorials? If no, how can I speed up video reading/decoding?

quentonh, 2025-02-14:
@mtntruong One option that you could evaluate and consider is the use of VVAS which supports hardware accelerated decoding on Zynq UltraScale+ devices, in conjunction with Vitis AI DPUs:  https://xilinx.github.io/VVAS/main/build/html/index.html

mtntruong, 2025-02-14:
@quentonh Thank you very much for this useful information. I have studied many Vitis libraries but this is the first time I have heard of VVAS. I will try integrating it into my project. Thanks again for you help.

closed 2025-04-08 by mtntruong