Xilinx/Vitis-AI #1513: Deploying YOLOv7 to VCK5000

yoyokst125, 2025-03-31:
I have successfully trained a YOLOv7s model with my custom dataset and completed the quantization and compilation processes (currently at the step using 'vai_c_xir -x /PATH/TO/quantized.xmodel -a /PATH/TO/arch.json -o /OUTPUTPATH -n netname'), which generated an .xmodel file. However, I'm having difficulty understanding the next steps for deploying and running the model on the VCK5000 platform after reviewing the reference manual. I would appreciate any guidance on this matter. Thank you.

mtntruong, 2025-04-08:
Hello,
To deploy YOLOv5, you can follow the instruction on using Vitis AI API_1 [here](https://docs.amd.com/r/en-US/ug1354-xilinx-ai-sdk/Developing-with-Vitis-AI-API_1). The instruction is for YOLOv3, but it is the same for YOLOv5, just change the C++ source files and prototxt files accordingly.

The C++ source code for YOLOv5 is available [here](https://github.com/Xilinx/Vitis-AI/tree/1eed93cde2ff3ed841ddb03dff0714b4e2d34f5b/examples/vai_library/samples/yolov5)

You can find the prototxt files for YOLOv5 in the .tar.gz files for vck5000 [here](https://github.com/Xilinx/Vitis-AI/blob/v3.0/model_zoo/model-list/pt_yolov5s6_coco_1280_1280_17G_3.0/model.yaml), [here](https://github.com/Xilinx/Vitis-AI/blob/v3.0/model_zoo/model-list/pt_yolov5-large_coco_640_640_109.6G_3.0/model.yaml), or [here](https://github.com/Xilinx/Vitis-AI/blob/v3.0/model_zoo/model-list/pt_yolov5-nano_coco_640_640_4.6G_3.0/model.yaml).

Hope this helps.

mtntruong, 2025-04-16:
@yoyokst125 "Unable to init server: Could not connect: Connection refused" is likely related to X11-forwarding. Testing with videos requires X11-forwarding to display the output videos. Is your VCK5000 card installed on a remote machine? If yes, you should enable X11-forwarding when connecting to the remote machine by using the command `ssh -X user@ip-address`

yoyokst125, 2025-04-16:
@mtntruong Thank you for your reply. I am currently using MobaXterm to remotely connect to a server equipped with VCK5000 for operations. After performing the following graphical operations and logging in

[image]

 I can see the graphical interface. After opening the terminal through the graphical interface and running the command sudo -E ./test_video_yolov5 yolov5_custom test.mp4 -t 8, I still receive the following message:

WARNING: Logging before InitGoogleLogging() is written to STDERR
I0416 03:53:20.676054 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:22.480751 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:24.228227 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:25.813521 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:27.535184 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:29.148936 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:30.655282 450 demo.hpp:754] DPU model size=640x640
I0416 03:53:32.206315 450 demo.hpp:754] DPU model size=640x640
Unable to init server: Could not connect: Connection refused

Is there a step I might have done incorrectly?

Since I am using MobaXterm for remote connection, should the command sudo apt-get install xauth be installed only on the VCK5000 server?

Thank you very much for your help.

mtntruong, 2025-04-17:
Hello, to use X11-forwarding, you need an X11 display server on your machine, not the remote. You have two options:
1. Switch to Linux (a virtual machine also works, but WSL may not) and connect to the server by `ssh -X user@ip-address`. Linux has a built-in X11 display server so you don't need to install anything (other than `xauth` as mentioned earlier if the distro does not have it by default).
2. If you can't switch to Linux, you need to install [Xming](http://www.straightrunning.com/XmingNotes/), either [free version](https://sourceforge.net/projects/xming/files/Xming/6.9.0.31/Xming-6-9-0-31-setup.exe/download) or [commercial license](http://www.straightrunning.com/XmingNotes/#head-16), then use your SSH client to connect with X11-forwarding enabled. I've used Xming with PuTTY and Bitvise, I'm not sure about MobaXterm. Note that the free version is 18 years old (but it works on Windows 10).

yoyokst125, 2025-04-17:
@mtntruong  Thank you for your reply. Currently, I am trying to run test_video_yolov5 directly on the server host (my server host is a GUI-enabled Ubuntu) without using remote connection, but I am still getting an error. Here is my running screen.

[image]

Thank you for your help!!!

mtntruong, 2025-04-17:
@yoyokst125 Ah.... so you are running the script inside a docker container. To use X11 inside Vitis container, add the following lines to the variable `docker_run_params` in [`docker_run.sh`](https://github.com/Xilinx/Vitis-AI/blob/1eed93cde2ff3ed841ddb03dff0714b4e2d34f5b/docker_run.sh#L75) (put them after `--network=host \`)
```
--pid=host \
-e DISPLAY=$DISPLAY \
-v /tmp/.X11-unix:/tmp/.X11-unix:ro \
```
The complete settings look like this
```
docker_run_params=$(cat <<-END
    -v /dev/shm:/dev/shm \
    -v /opt/xilinx/dsa:/opt/xilinx/dsa \
    -v /opt/xilinx/overlaybins:/opt/xilinx/overlaybins \
    -e USER=$user -e UID=$uid -e GID=$gid \
    -v $DOCKER_RUN_DIR:/vitis_ai_home \
    -v $HERE:/workspace \
    -w /workspace \
    --rm \
    --network=host \
    --pid=host \
    -e DISPLAY=$DISPLAY \
    -v /tmp/.X11-unix:/tmp/.X11-unix:ro \
    ${DETACHED} \
    ${RUN_MODE} \
    $IMAGE_NAME \
    $DEFAULT_COMMAND
END
)
```
Then use this modified `docker_run.sh` to start the container.

yoyokst125, 2025-04-19:
@mtntruong It has successfully run. Thank you for your help!!!